realthunder/fcad_pcb #4: pcb.makeCopper(fuseCoppers=True) fails with some new pads

easyw, 2018-03-31:
Hi, 
following our recent conversation I'm uploading some boards with the following pads:
- round-rect pads
[rounded-rect-pcb.zip](https://github.com/realthunder/fcad_pcb/files/1865561/rounded-rect-pcb.zip)

- polylines
[polyline-pcb.zip](https://github.com/realthunder/fcad_pcb/files/1865560/polyline-pcb.zip)

Moreover I'm attaching also a very simple case in which the pcb.make() works, but the pcb.makeCopper(fuseCoppers=True) not
[pads-board.zip](https://github.com/realthunder/fcad_pcb/files/1865562/pads-board.zip)
[pads-drilled-board.zip](https://github.com/realthunder/fcad_pcb/files/1865563/pads-drilled-board.zip)
(please note that your test.kicad_pcb works jsut fine, also if saved in kicad v5)

Thx in advance
Maurice

realthunder, 2018-05-01:
I have fixed issues shown in your files. And added round rect and polyline pad. I notice that the new KiCAD allows other primitive geometry types in custom pad. If you want those, please attach a file here, and I'll add it.

realthunder, 2018-05-01:
Oh, just in case. You title is kind of wrong. 'fuseCoppers' argument is for make() method, not makeCopper(). BTW, I've updated readme file to show how to make just the shape without adding document feature.

easyw, 2018-05-01:
Hi @realthunder 
Thx for going into this...
I noticed that for roundrect board ([rounded-rect-pcb.zip](https://github.com/realthunder/fcad_pcb/files/1865561/rounded-rect-pcb.zip)) there is an issue:
`Exception (Tue May 01 10:07:49 2018): shapes are not planar  `
when running `pcb.make()`
and it doesn't create the F.Cu with this command
`pcb.makeCopper(holes=True)  `
and with this
`pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True )`

> I notice that the new KiCAD allows other primitive geometry types in custom pad. If you want those, please attach a file here, and I'll add it.

I just noticed them too ... I'll add a testing board... thx again
Maurice

realthunder, 2018-05-01:
Just fixed. Please try again.

easyw, 2018-05-01:
those are fine now, thx 😄 
I'm adding an other case in which the
`pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True )`
is failing... with a NPTH pad
here a board test case...
[test-NPTH.zip](https://github.com/realthunder/fcad_pcb/files/1963791/test-NPTH.zip)

realthunder, 2018-05-01:
Fixed. This is caused by tolerance problem when cutting a round pad with a NPTH with the exact same size. So I enlarge all NPTH holes size by the same amount as the copper thickness. Is that okay?

easyw, 2018-05-02:
Hi @realthunder 
I still get an error on creation with test-NPTH.kicad_pcb and this command:
`pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True )`
and
`pcb.makeCopper(holes=True) `

realthunder, 2018-05-02:
It works fine on my PCB. What version of FreeCAD are you using? Can you please show me the error message? And please try to set a larger size offset and try again.

```
pcb.hole_size_offset = 0.005
```