realthunder/fcad_pcb #6: No module named future.utils

traveler787, 2018-06-15:
I am getting an error from kicad.py line 4.  It says "No module named future.utils"

I cloned the repo to C:\Users\30920\AppData\Roaming\FreeCAD\Macro\fcad_pcb
and checked out the submodules.

Here is my contents of fcad_pcb
----------------
.git
kicad_parser
screenshots
tests
.gitignore
.gitmodules
__init__.py
__init__.pyc
kicad.py
kicad.pyc
LICENSE
Readme.md
----------------

Any help would be appreciated.
Brian

realthunder, 2018-06-16:
Which platform are you in? Future is a standard python module, and can be installed with pip.

traveler787, 2018-06-19:
I am on Windows 10.  I suspected that was what I needed.  With some more searching I found how to install the Future module...

Open an administrative command prompt at the FreeCAD bin directory (ie.  C:\Program Files\FreeCAD 0.17\bin)
Run "python -m pip install future"

Is the script able to generate solder mask or silk screen layers, or load 3D components?

Very nice script.  Thank you.

realthunder, 2018-06-20:
Solder mask and silk screen are not supported yet. Loading of 3D components is probably broken with the current version of FreeCAD. You can try the [StepUp](https://www.freecadweb.org/wiki/KicadStepUp_Workbench/it) workbench. You can install it using FreeCAD addon manager.

traveler787, 2018-06-20:
Thank you again.

closed 2018-06-20 by traveler787