realthunder/fcad_pcb #14: Option to import only copper with specific netname

MitjaNemec, 2019-03-25:
I am currently using your module for DC power analysis using FreeCad FEM solver. I've pretty much got it covered on small test board, but I can see the scalability issue. As the results are needed only for specific nets, it would be great if import would have an option to specify which net to import only. This should significantly reduce the complexity for the mesh and consequently FEM solver.

Or you could import the whole board, but in such a way that specific nets could be modified/deleted.

realthunder, 2019-03-26:
Implemented. Checkout the usage [here](../#user-content-net-filter)

closed 2019-03-26 by realthunder

MitjaNemec, 2019-03-26:
Much appreciated. I am already testing it and I've found an issue, which is more a KiCad issue than fcad_pcb issue.

The nets which derive the name from local labels in schematics have slash prepended even if they are on root of the hierarchy. If they are deeper in hierarchy they have the full schematics hierarchy prepended. But if the netnames are generated by KiCad (format is: net-(###-###)) they have no hierarchy prepended. So you might just add a note to readme so that the users don't get tripped up when deriving netnames from schematics.

Reporting which net was not found is a great feature so that you find the issue quickly

commit a11b7ac4 (2019-03-26): Support filter by net name

Closes #14