realthunder/fcad_pcb #17: New freecad user, trying to figure out how to input correctly

richard-phat-the-nguyen, 2019-05-13:
Hi,
I am a new user to freecad and I am having trouble with the syntax.
I have a file located at C:\Users\richa\Desktop\ODS Med\Ring_PCB\Ring_PCB.kicad_pcb

When I type the following I get some error messages below. I am not sure where I messed up.

```
from fcad_pcb import kicad
pcb = kicad.KicadFcad(<C:\Users\richa\Desktop\ODS Med\Ring_PCB\Ring_PCB.kicad_pcb>)
pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True )
```

> Traceback (most recent call last):
>   File "C:\Program Files\FreeCAD 0.18\bin\lib\codeop.py", line 168, in __call__
>     return _maybe_compile(self.compiler, source, filename, symbol)
>   File "C:\Program Files\FreeCAD 0.18\bin\lib\codeop.py", line 99, in _maybe_compile
>     raise err1
>   File "C:\Program Files\FreeCAD 0.18\bin\lib\codeop.py", line 87, in _maybe_compile
>     code1 = compiler(source + "\n", filename, symbol)
>   File "C:\Program Files\FreeCAD 0.18\bin\lib\codeop.py", line 133, in __call__
>     codeob = compile(source, filename, symbol, self.flags, 1)
>   File "<input>", line 1
>     pcb = kicad.KicadFcad(<C:\Users\richa\Desktop\ODS Med\Ring_PCB\Ring_PCB.kicad_pcb>)
>                           ^
> SyntaxError: invalid syntax
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "C:\Program Files\FreeCAD 0.18\bin\lib\codeop.py", line 168, in __call__
>     return _maybe_compile(self.compiler, source, filename, symbol)
>   File "C:\Program Files\FreeCAD 0.18\bin\lib\codeop.py", line 70, in _maybe_compile
>     for line in source.split("\n"):
> SystemError: <built-in method split of str object at 0x0000021300821670> returned a result with an error set
> 
> The above exception was the direct cause of the following exception:
> 
> SystemError: <built-in method split of str object at 0x0000021300821530> returned a result with an error set

realthunder, 2019-05-13:
You are quoting the path string wrongly. Should be like this,
```python
pcb = kicad.KicadFcad('C:/Users/richa/Desktop/ODS Med/Ring_PCB/Ring_PCB.kicad_pcb')
```
Notice the backslash changing, because otherwise it will be treated as escaped character. You can sort of disable the escapes by proceeding the quote with `r`, like this
```python
pcb = kicad.KicadFcad(r'C:\Users\richa\Desktop\ODS Med\Ring_PCB\Ring_PCB.kicad_pcb')
```

richard-phat-the-nguyen, 2019-05-13:
Hi,
following the modification, I still have trouble to run it.:

Just to make sure if I haven't messed up anything,  I have the macro installed at:
C:\Users\richa\AppData\Roaming\FreeCAD\Macro\Mod\fcad_pcb 
and the macro directory is set at 
C:/Users/richa/AppData/Roaming/FreeCAD/Macro/Mod

Once copy paste the following code, I get the following message below.

```
from fcad_pcb import kicad
pcb = kicad.KicadFcad('C:/Users/richa/Desktop/ODS Med/Ring_PCB/Ring_PCB.kicad_pcb')
pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True )
```

```
>>> from fcad_pcb import kicad
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:/Users/richa/AppData/Roaming/FreeCAD/Macro/Mod\fcad_pcb\kicad.py", line 18, in <module>
    from .kicad_parser import KicadPCB,SexpList
ImportError: cannot import name 'KicadPCB'
>>> pcb = kicad.KicadFcad('C:/Users/richa/Desktop/ODS Med/Ring_PCB/Ring_PCB.kicad_pcb')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'kicad' is not defined
>>> pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True )
```

realthunder, 2019-05-13:
You didn't download the git submodules. If you are cloning my git repo, go to `C:\Users\richa\AppData\Roaming\FreeCAD\Macro\Mod\fcad_pcb` and run
```shell
git submodule update --init --recursive
```
If you are downloading the zip archive from github, then please go to the [release](../releases) page and download the `fcad_pcb.zip` file there, extract and replace the existing directory.

richard-phat-the-nguyen, 2019-05-13:
Thanks,
I download as you instructed and now it works just fine. The generated step file let me to have much more control when I am creating the cam.

[image]

vaibhavmittal001, 2019-07-05:
were do i need to type theese commands
from fcad_pcb import kicad
pcb = kicad.KicadFcad(<C:\Users\richa\Desktop\ODS Med\Ring_PCB\Ring_PCB.kicad_pcb>)
pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True )

realthunder, 2019-07-05:
Just look at the picture above. The lower half of the window is called `Python console`. Type in there. If you don't see it, you can reveal it through menu `View -> Panels -> Python console`. 

Make sure you use backslash for the path, e.g.

```
pcb = kicad.KicadFcad('C:/Users/richa/Desktop/ODS Med/Ring_PCB/Ring_PCB.kicad_pcb')
```

rniueng, 2020-02-18:
For a new user who doesn't know how to run Freecad Macros and how to update git submodules, the installation instruction is not clear. I have the same problem as shown in below when I tried to run the first line.

Python 3.6.6 | packaged by conda-forge | (default, Jul 26 2018, 11:48:23) [MSC v.1900 64 bit (AMD64)] on win32
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>>  from fcad_pcb import kicad
  File "<input>", line 1
    from fcad_pcb import kicad
    ^
IndentationError: unexpected indent
>>> 

When execute the kicad.py in Macro window manually, it shew the similar error message at linen 18 like:
  File "C:/Users/richa/AppData/Roaming/FreeCAD/Macro/Mod\fcad_pcb\kicad.py", line 18, in <module>
    from .kicad_parser import KicadPCB,SexpList

realthunder, 2020-02-18:
You probably didn't clone all the submodules. You download the prepackaged [fcad_pcb.zip](https://github.com/realthunder/fcad_pcb/releases/download/1.1/fcad_pcb.zip), and extract it to your FreeCAD Macro directory. Please delete your previous clone of fcad_pcb first.

When you copy the Python commands, make sure to not include any leading space.

rniueng, 2020-02-19:
Thank you realthunder! Let me describe my questions in a new user perspective. Hope these questions may help other guys later.

Firstly, I noticed in the instructions, it mentioned the version is 0.17. I'm using 0.18. I tried the 0.17 Portable version, it will shown a error on line 4 instead.

Regarding how to install:
1. I used GitHub desktop and fetched this rep. Then copied the fcad_pcb folder in my GitHub local folder to the target "C:/Users/xxx/AppData/Roaming/FreeCAD/Macro/Mod/fcad_pcb" folder.
2. There is a question, there's no "Mod" subfolder in Macro folder after FreeCAD is installed. I created the Mod subfolder myself. Hope this doesn't impact any file path variable in execution.
3. When got problem, I also downloaded the fcad_pcb.zip file found in another discussion. And either way, the problem is same.
4. Even I tried the GitGui toll for windows, and did clone the repo also executed "git submodule update --init --recursive" in Git Bash terminal window. same problem.

The Macro execution:
1. Can I specify the Macro folder to any other folder? As asked above, not sure if can create a subfolder Mod in default Macro folder manually.
2. Should I new a document firstly before execute the Macro. Looks the error message is same.
3. Please advise if my operation is correct by clicking the Execute button below after select the kicad.py?
[image]
4. The error message is shown as below after execute the kicad.py:
[image]

Thanks for your help!

realthunder, 2020-02-20:
> * I used GitHub desktop and fetched this rep. Then copied the fcad_pcb folder in my GitHub local folder to the target "C:/Users/xxx/AppData/Roaming/FreeCAD/Macro/Mod/fcad_pcb" folder.
> * There is a question, there's no "Mod" subfolder in Macro folder after FreeCAD is installed. I created the Mod subfolder myself. Hope this doesn't impact any file path variable in execution.

Exact `fcad_pcb.zip` file, and copy `fcad_pcb` directory directly into the `Macro` directory, not `Mod`. This shall solve the rest of the problems. Try run the `from fcad_pcb import kicad` command (without leading space) in FreeCAD console.

rniueng, 2020-02-20:
Okay. Now I have figured out the problem. I should not try to execute the .py script file manually in the Macro window. To executed the script file, I had to change the default path of user local macro setting to the sub-folder fcad_pcb. All these lead to my problem. So the hints for new user are:
1. Just put the fcad_pcb clone to the /FreeCAD/Macro folder
2. Don't change the default User Macros location which should be C:/Users/xxxx/AppData/Roaming/FreeCAD/Macro"
3. Should restart the FreeCAD after copy the files to Macro folder.
The other hint for the file format in pcb = kicad.KicadFcad(<full_path_to_your_kicad_pcb_file>). I think it will be more clear if change it to: pcb = kicad.KicadFcad('full_path_to_your_kicad_pcb_file').

Thanks again for the help!

Rerouter, 2020-03-01:
Hello, been working through trying to get this tool working, but it seems like it doesn't like something about my PCB file (seems to have an issue with some custom pad?)

```
>>> pcb.setNetFilter('-5v')
>>> pcb.makeCopper()
Traceback (most recent call last):
  File "C:/Users/User/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 272, in make_custom
    pts = params.primitives.gr_poly.pts
  File "C:/Users/User/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad_parser\sexp_parser\sexp_parser.py", line 168, in __getattr__
    raise AttributeError('{}: key "{}" not found'.format(self._line,name))
AttributeError: 9202: key "gr_poly" not found

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:/Users/User/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1615, in makeCopper
    thickness=thickness)
  File "C:/Users/User/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1302, in makePads
    w = make_shape(Vector(*p.size),p)
  File "C:/Users/User/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 279, in make_custom
    raise RuntimeError('Cannot find polyline points in custom pad')
RuntimeError: Cannot find polyline points in custom pad
```
[example pcb.zip](https://github.com/realthunder/fcad_pcb/files/4271746/example.pcb.zip)

As I remain unclear on if it is user error like the above or a seperate issue, I am putting it here for now. some nets, e.g. GNDS, GND work, but no others. 
[image]

easyw, 2020-03-01:
> RuntimeError: Cannot find polyline points in custom pad

probably the issue is related to your TO-4-4 footprint... you designed it with some geometry pads, but you also included a simple track in it, not creating a real pad.
here the fp and the fp modified having all pads as geometry pads.
[TO-4-4-modified.zip](https://github.com/realthunder/fcad_pcb/files/4271999/TO-4-4-modified.zip)

realthunder, 2020-03-09:
> Hello, been working through trying to get this tool working, but it seems like it doesn't like something about my PCB file (seems to have an issue with some custom pad?)

The problem is because of the arc in side the custom pad. I have just add support for that. Please sync my repo and try again.

easyw, 2020-03-09:
@realthunder 
would you mind to try the modified file?
[High.Resolution.Multislope.converter-modified.kicad_pcb.zip](https://github.com/realthunder/fcad_pcb/files/4272012/High.Resolution.Multislope.converter-modified.kicad_pcb.zip
)
I have modified the module internal pad done as track (which shouldn't be used), with a circle geometry custom pad instead... (primitives, gr_circle)
With this file I still get errors:
> Traceback (most recent call last):
>   File "/home/mau/Cad/fcad_pcb-test.py", line 81, in <module>
>     pcb.makeCopper(holes=True)
>   File "/home/mau/.FreeCAD/Macro/fcad_pcb/kicad.py", line 1630, in makeCopper
>     thickness=thickness)
>   File "/home/mau/.FreeCAD/Macro/fcad_pcb/kicad.py", line 1315, in makePads
>     w = make_shape(Vector(*p.size),p)
>   File "/home/mau/.FreeCAD/Macro/fcad_pcb/kicad.py", line 286, in make_custom
>     logger.warning('Unknown primitive {} in custom pad', key)
> 
.
[image]

realthunder, 2020-03-10:
Fixed. Please check.

easyw, 2020-03-10:
fix confirmed!
thx :smiley_cat:

linuxcf, 2020-05-12:
This is not really an issue, but I am posting it here as a question/request.  What if I want to convert to STEP a pcb with _several_ layers, not just two, of _varying_ dielectric thickness? Ditto for the copper weight on every layer.  I guess it would be possible to import the pcb into FreeCAD two layers at a time and then align all of them in FreeCAD, but I thought it would be easier to do it all at once with your script.  Also, due to lack of available documentation, I am not sure what the options 'combo' and 'fuseCoppers' do.

I would like to report that, following your examples, I have used your macro successfully with FreeCAD 0.19 (development) on stock Ubuntu 16.04LTS, although you do not mention FreeCAD 0.19.

realthunder, 2020-05-14:
> This is not really an issue, but I am posting it here as a question/request. What if I want to convert to STEP a pcb with _several_ layers, not just two, of _varying_ dielectric thickness?

I just added this feature. With `pcb.make()`, you can now pass in a dictionary for `copper_thickness`. I have also updated the Readme for instruction, copied below

> supply copper thickness per layer, pass a dictionary instead. Use either integer (0~31, 0 being the front and 31 the back), or layer name for key. Key None can be used for default thickness.

```python
pcb.make(copper_thickness={None:0.05, 0:0.04, 'B.Cu':0.09}, board_thickness=1.53, combo=False, fuseCoppers=True)
```

MitjaNemec, 2020-05-14:
I think that @linuxcf also requested for user to have the control of thicknes of dielectric layers, not just the copper layers.

It might make more sense to implement parser for the board stackup info that is present in the 5.99 file format. So even 5.1.x users could design the pcb. and then reopen it in 5.99, define the stackup, and then import the .step into FreeCad via `fcad_pcb`

example of stackup info in 5.99:
```
(stackup
      (layer "F.SilkS" (type "Top Silk Screen"))
      (layer "F.Paste" (type "Top Solder Paste"))
      (layer "F.Mask" (type "Top Solder Mask") (color "Green") (thickness 0.01))
      (layer "F.Cu" (type "copper") (thickness 0.035))
      (layer "dielectric 1" (type "core") (thickness 0.01) (material "FR2") (epsilon_r 1) (loss_tangent 0))
      (layer "In1.Cu" (type "copper") (thickness 0.035))
      (layer "dielectric 2" (type "prepreg") (thickness 1.41) (material "FR4") (epsilon_r 4.5) (loss_tangent 0.02))
      (layer "In2.Cu" (type "copper") (thickness 0.035))
      (layer "dielectric 3" (type "core") (thickness 0.01) (material "FR4") (epsilon_r 4.5) (loss_tangent 0.02))
      (layer "B.Cu" (type "copper") (thickness 0.035))
      (layer "B.Mask" (type "Bottom Solder Mask") (color "Green") (thickness 0.01))
      (layer "B.Paste" (type "Bottom Solder Paste"))
      (layer "B.SilkS" (type "Bottom Silk Screen"))
      (copper_finish "None")
      (dielectric_constraints no)
    )
```

MitjaNemec, 2020-05-14:
If I was not clear enough, this would require also to have separate board objects/parts (substrate, prepreg, ...)

realthunder, 2020-05-14:
> It might make more sense to implement parser for the board stackup info that is present in the 5.99 file format. So even 5.1.x users could design the pcb. and then reopen it in 5.99, define the stackup, and then import the .step into FreeCad via `fcad_pcb`

Can you please provide a complete kicad_pcb file with that stackup info?

MitjaNemec, 2020-05-14:
Board stackup dialog:
[image]

layout file:
[nightly_test.zip](https://github.com/realthunder/fcad_pcb/files/4627237/nightly_test.zip)

linuxcf, 2020-05-14:
@MitjaNemec and @realthunder:  Thank you for your hard work and information!

The reason I asked the question about thicknesses is new(er) techniques of so-called substrate integrated waveguides (SIW), where, basically, you create waveguides with "soft" walls comprised of vias.  The "soft" waveguides may be parallel to the layers, as in [this wikipedia article](https://en.wikipedia.org/wiki/Substrate_integrated_waveguide) or normal to the dielectric layers, as in [this article about SIW antennas](https://www.researchgate.net/publication/319543947_A_Review_Substrate_Integrated_Waveguide_Antennas_and_Arrays).  Such structures need a full 3D electromagnetics simulator and the easiest way to pass data to such a program, open-source or commercial, is a STEP file.  (Unfortunately, closed-source 3D electromagnetics simulators hold a huge lead over their open-source counterparts, but that's another topic.)  The interest in SIW is intensified today because you can build phased arrays inexpensively on pcbs and phased arrays can (mostly) replace mechanically-steered antennas when tracking low-Earth orbit satellites, which are proliferating today.  I am writing this to show that the paradigm of pcb design is shifting from the traditional 2D to 3D.  There is no reason why KiCAD cannot be on the forefront of such developments and this macro does exactly that for KiCAD.

In a SIW the copper structures may span several pcb layers.  If I can accurately export the copper structures to a 3D electromagnetics simulator, I can fill in the dielectrics in the padstack by hand.  So, even exporting accurately the copper structures alone would be great help.

I looked at the wiki page for this project, but was not able to find documentation about the 'combo' and 'fuseCopper' options.

realthunder, 2020-05-15:
> I looked at the wiki page for this project, but was not able to find documentation about the 'combo' and 'fuseCopper' options.

`combo` will cause `make()` to create a Compound object that include everything. This is probably not what you want. `fuseCopper` will cause all layers of copper solid to fuse together and connected them with any tubes from plated through holes.

Tell me anything you want exactly, preferably with demo files. I'll see what I can do.

linuxcf, 2020-05-15:
@realthunder:  Thank you so much!  I will go with 'fuseCopper=True' and 'combo=False'.

linuxcf, 2020-05-20:
@realthunder: I made the attached pcb, using via fencing, via stitching, and teardrops (thank you, @easyw!).  Stupid question:  The intended copper thickness was 35μm, but FreeCAD measures 35.0001μm.  I know that in practice sub-nm tolerances are meaningless, but I wonder whether this error could pile up, perhaps nonlinearly, if I had multiple layers?  Just a question.

I think this macro deserves much wider publicity.

[layout-to-step.zip](https://github.com/realthunder/fcad_pcb/files/4657806/layout-to-step.zip)

iCarletto, 2020-08-05:
Hi, I'm having a similar issue to @Rerouter. After loading my pcb [Bus.zip](https://github.com/realthunder/fcad_pcb/files/5028770/Bus.zip) and issuing the basic command

`pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True )`

I get the following message
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Users\carlo\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad.py", line 1965, in make
    board_thickness=board_thickness)
  File "C:\Users\carlo\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad.py", line 1802, in makeCoppers
    holes=hole_shapes,z=z,prefix=None,fuse=fuse)
  File "C:\Users\carlo\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad.py", line 1721, in makeCopper
    thickness=thickness)
  File "C:\Users\carlo\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad.py", line 1611, in makeZones
    if unquote(z.layer) != self.layer or self.filterNets(z):
  File "C:\Users\carlo\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad_parser\sexp_parser\sexp_parser.py", line 168, in __getattr__
    raise AttributeError('{}: key "{}" not found'.format(self._line,name))
AttributeError: 8909: key "layer" not found
```

Is this related to my pcb?

realthunder, 2020-08-07:
> Hi, I'm having a similar issue to @Rerouter. After loading my pcb [Bus.zip](https://github.com/realthunder/fcad_pcb/files/5028770/Bus.zip) and issuing the basic command

Problem fixed. Please sync your repo.

BTW, this PCB is quite complex. It took more than 15 minutes and a few gigabytes of memory to make the fused copper on my computer.

iCarletto, 2020-08-07:
Hi @realthunder, thank you very much. That's a great tool!
I actually did not need the entire board but only a specific net for further FEM simulation, so I used the respective command setNetFilter. The created solid looks really cool, but I'm not able to generate a mesh either within FreeCAD or outside using the step file. Do you have any hint in this regard?

realthunder, 2020-08-09:
Which net did you try? Any error message?

MitjaNemec, 2020-08-11:
@iCarletto I had best result using standolne gmesh. You can make the tracks thicker and account for this in specific conduction/resistance (thermal or electrical)

MitjaNemec, 2020-08-11:
@iCarletto also simplifed vias (#22) help a lot.

iCarletto, 2020-08-11:
@realthunder I'm interested in net VBUS_C (98). When I try to generate the mesh in FreeCAD using Gmsh I receive the error `Error : Invalid boundary mesh (segment-facet intersection) on surface 3106`. I will try to use the suggestion from @MitjaNemec, however I would prefer not to simplify the vias because my objective is to check whether the current density is omogeneous through the vias.

I tried to export the net layer as step file and to import it in Comsol (middle image), but I receive a similar error when trying to build the mesh because of some intersecting faces (lower image). If I reduce the absolute import tolerance (upper image) from the standard 1e-5 to 5e-5, it succed but as a consequence the edges of the copper layer collapse in a line and I'm not sure what happens to thickness of the layer in the center.

I guess the problem of intersecing faces could be due to either the accuracy of the exported step file, or the way vias are built, but honetsly I'm using FreeCAD for the first time and I do not fully understand most of the parameters. Do you think there might some essential parameter that I'm overlooking?

[image]

MitjaNemec, 2020-08-14:
@iCarletto meshin thin hollow cylinders is a PITA. 

And if you wish to get accurate results you'll need to talk to PCB manufacturer regarding the via thickness it is certainly not the same as track thick thickness. And when you're there you can simplify vias and set them up for precise copper area you want

luzpaz, 2022-10-30:
Can this ticket be closed ?

electr0mancer, 2024-07-05:
Hi, I am getting the following errors after using the following lines of code
code:
from fcad_pcb import kicad
pcb=kicad.KicadFcad( 'C:\FEA\Thermal\Thermal.kicad_pcb')
pcb.setNetFilter('"GND"')
pcb.make(copper_thickness=0.070, board_thickness=1.46, combo=False, fuseCopper=True)

Error:

>>> from fcad_pcb import kicad
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'fcad_pcb'
>>> pcb=kicad.KicadFcad( 'C:\FEA\Thermal\Thermal.kicad_pcb')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'kicad' is not defined
>>> pcb.setNetFilter('"GND"')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'pcb' is not defined
>>> pcb.make(copper_thickness=0.070, board_thickness=1.46, combo=False, fuseCopper=True)
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'pcb' is not defined