realthunder/fcad_pcb #20: Trapezoid pads not yet implemented

Gasman2014, 2020-02-10:
Running this against a recent board using trapezoidal pads produces the following error. Otherwise seems to work well.

```
File "/Users/xyz/Library/Preferences/FreeCAD/fcad_pcb/kicad.py", line 1181, in makePads
    'pad shape {} not implemented\n'.format(shape))
NotImplementedError: pad shape trapezoid not implemented
```

realthunder, 2020-02-10:
Can you please attach a file with this type of pad?

Gasman2014, 2020-02-10:
Component U1 (MAR-6) Amplifier on the attached board uses a footprint which requires a number of trapezoidal pads. I was hoping to do some thermal modelling of this.

[1GHz Elektor Probe.kicad_pcb.zip](https://github.com/realthunder/fcad_pcb/files/4183542/1GHz.Elektor.Probe.kicad_pcb.zip)

Gasman2014, 2020-02-10:
Have highlighted one of the component pads showing how this shaped pad is made up of several tessellated trapezoidal sub-pads.

[image]

realthunder, 2020-02-10:
Thanks. I'll add it soon.

realthunder, 2020-02-11:
Done.

closed 2020-02-11 by realthunder

Gasman2014, 2020-02-11:
Nice, thank you! Very speedy service. :)
[image]

easyw, 2020-02-12:
@realthunder 
I still have an issue with a trapezoidal footprint coming from kicad wizard ...
[touch-slider.zip](https://github.com/realthunder/fcad_pcb/files/4191637/touch-slider.zip)

Trapezoidal are not often used, compared to polyline pads though...

realthunder, 2020-02-13:
Fixed

easyw, 2020-02-13:
@realthunder 
I'm still having a little issue... can you please confirm it?
<code>
pcb.setLayer('F.Cu');pcb.makeCopper(holes=True)
</code>
[image]

realthunder, 2020-02-13:
It works on my side. Can you show me your FC about info? Also please save the FCStd file and attached here.

easyw, 2020-02-13:
> OS: Windows 10 (10.0)
> Word size of OS: 64-bit
> Word size of FreeCAD: 64-bit
> Version: 0.19.19518 (Git)
> Build type: Release
> Branch: master
> Hash: 16b2b9a864fbf8fcae6af1e38640995fd0fefeaf
> Python version: 3.8.1
> Qt version: 5.12.5
> Coin version: 4.0.0
> OCC version: 7.4.0
> 

> OS: Windows 10
> Word size of OS: 64-bit
> Word size of FreeCAD: 64-bit
> Version: 0.18.4 (GitTag)
> Build type: Release
> Branch: releases/FreeCAD-0-18
> Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
> Python version: 3.6.6
> Qt version: 5.6.2
> Coin version: 4.0.0a
> OCC version: 7.3.0
> 

> OS: Windows 10 (10.0)
> Word size of OS: 64-bit
> Word size of FreeCAD: 64-bit
> Version: 0.19.19563 (Git)
> Build type: Release
> Branch: master
> Hash: b799996c22634adfe246723bf0eaa96804e926ce
> Python version: 3.6.8
> Qt version: 5.12.1
> Coin version: 4.0.0a
> OCC version: 7.3.0
> 

@realthunder 
[touch-slider-FC.zip](https://github.com/realthunder/fcad_pcb/files/4200079/touch-slider-FC.zip)

realthunder, 2020-02-14:
Updated. Please try again. It it still happens, please try set the following parameter and try again.

```
pcb.pad_inflate = 0.0001
```

easyw, 2020-02-14:
> Updated. Please try again.

@realthunder 
fix confirmed!
Thx again 😸

commit ccd32c9c (2020-02-11): Add support for trapezoid pad

Fixes #20