realthunder/fcad_pcb #22: Simplify vias

MitjaNemec, 2020-02-12:
Hi! 

Would it be possible to add the option with which vias (hollow cylinder with thin wall) would be substituted with full rectangular object with the same crosssection. Even better would be if crosssection could be controlled.

I am tinkering with FEM simulation of a PCB (thermal and electrostatic) and vias require dense meshing. This simplification would caous significantly less meshing nodes thus reducing FEM calculatio time

realthunder, 2020-02-12:
Done.

You can control the via simplification using

```
pcb = kicad.KicadFcad(<filename>)
# -1 to disable via, 0 as normal, >0 as ratio to radius for the square
pcb.via_bound = 1
```

closed 2020-02-12 by realthunder

MitjaNemec, 2020-02-12:
First, thank you for such a prompt response. 

But it does not work as I would expect. There is still a round hole on the top/bot copper and there is no connection between the planes. As can be seen in following figure
[image]

I think we had a missunderstanding. Your solution just changes the solder paste on the copper layers.

If I import the board with `pcb.make(copper_thickness=0.20, board_thickness=1.4, combo=False, fuseCoppers=True )` as I want to in order to do a proper FEM analysis I still get via as a hollow cylinder connecting two copper layers. Instead of thy hollow cylinder I'd like to get square pillar and no hole on the copper layers. Is this doable?

realthunder, 2020-02-13:
Added new parameter 'via_skip_hole', set to True by default. It controls whether to skip hole if via_bound is set. So if you still want a square hole in the copper, you can set it to False.

MitjaNemec, 2020-02-13:
Hi, this is progress and I'd certainly like to keep this behaviour. But either, I can not express clearly what I want or there is a bug in the implementation.

As you can see in the pictures round hollow via is substituted with rectangular hollow pilar.
Could you add the option to make the rectangular pilar full. This should reduce the mesh complexity even further.

[image]

[image]

realthunder, 2020-02-13:
Can you show me your commands? I think the following commands will produce what you want,

```
pcb = kicad.KicadFcad('1GHz Elektor Probe.kicad_pcb', via_bound=1)
pcb.make(copper_thickness=0.20, board_thickness=1.4, combo=False, fuseCoppers=True )
```
What I mean in my last comment is that you don't need to set via_skip_hole if you don't want the the hole. It is by default 'True' if via_bound is set.

[image]

MitjaNemec, 2020-02-13:
Hi, sorry it looks like it was a PBKAC. I can confirm that it is working.

Again sorry for the noise and thank you for your support

realthunder, 2020-02-13:
No problem. Glad it works!

commit 85a6921b (2020-02-12): Add 'via_bound' parameter for via simplication

Closes #22