realthunder/fcad_pcb #23: Add support for oval holes; Add support for blind/micro vias

MitjaNemec, 2020-04-13:
Hi!

I really appreciate your work done so far. So I humbly ask if you could:
1. Add support for oval holes
2. Add support for micro/blind vias

Thanks in advance

I am attaching a test project with the screenshot how the macro behave currently (tested with 5cd7951c51767cb5831e33420846055d10454d62 on KiCad 5.1.5-90-g5c7ac1bb7(current nightly) an FreeCad 18.4.
Executed code:
``` python
from fcad_pcb import kicad
pcb = kicad.KicadFcad("D:/Mitja/Plate/ACS_fcad_pcb_test/ACS_fcad_pcb_test.kicad_pcb")
pcb.make(copper_thickness=0.070, board_thickness=1.32, combo=False, fuseCoppers=True )
```

KiCad 3D view, board and soldermask not shown:
[image]
FreeCad 3D view, board hidden:
[image]

KiCad project
[ACS_fcad_pcb_test.zip](https://github.com/realthunder/fcad_pcb/files/4469295/ACS_fcad_pcb_test.zip)

realthunder, 2020-04-14:
Added, please check. BTW, oval hole should be working with you current version. Not sure why it didn't work in your place. Can you please download the latest FC 0.19 dev and try?

closed 2020-04-14 by realthunder

MitjaNemec, 2020-04-15:
I can confirm that blind/micro vias work. But the oval hole is not working. Neither in 0.18.4 nor in 0.19.20514-Win-Conda_vc14.x-x86_64
[image]

easyw, 2020-04-15:
@realthunder 
I confirm to have the same issue with the test file provided.

realthunder, 2020-04-15:
Should be fixed by now. You'll need to update the submodules recursively with
```
git submodule update --recursive
```

MitjaNemec, 2020-04-15:
Confirming. Works as it should.

Again thank you for your support.

easyw, 2020-04-15:
+1
thanks 😺

commit d9149d95 (2020-04-14): Add support for blind vias

Closes #23