realthunder/fcad_pcb #24: missing support for (recently added) arc in pcb traces

easyw, 2020-04-14:
KiCAD nigtly finally has added support for round traces (not with a direct editing ATM, but importing from an external pcb format)
[pcbnew/pcb_parser.cpp#L3402](https://github.com/KiCad/kicad-source-mirror/blob/e93684015c0415b093e38761525a10306f8aff6b/pcbnew/pcb_parser.cpp#L3402)
the format is as following:
`(arc (start 110.538233 90.718234) (mid 109.913928 91.652573) (end 109.694701 92.754701) (width 0.199999) (layer "B.Cu") (net 1) (tstamp 70a20bf4-2890-460e-9f59-39c9ba6d66b4))`
I'm attaching a sample pcb:
[curved-traces.zip](https://github.com/realthunder/fcad_pcb/files/4473434/curved-traces.zip)

realthunder, 2020-04-14:
Support added. Please check.

realthunder, 2020-04-14:
BTW, you need to update kicad_parser sub module as well.

closed 2020-04-14 by realthunder

easyw, 2020-04-14:
Confirmed working!
Thx a lot :smile:

commit efc83300 (2020-04-14): Support 'arc' type PCB trace

Closes #24