realthunder/fcad_pcb #25: Missing oval pads

9robs, 2020-05-05:
Commenting based on this KiCAD forum thread:
https://forum.kicad.info/t/export-only-trace-vias-as-3d-model/22683/5

I LOVE this tool - thank you so much for developing it...but in a board I am trying to import it seems the oval pads are not being created...Any hints? What am I missing?

[image]

[image]

realthunder, 2020-05-05:
Is is possible to provide a sample file for this problem?

9robs, 2020-05-06:
Attached - there is a lot going on with this board and it is not anything 'real'...just more for R&D purposes (imported from Eagle)....if you would like a simpler version with the same issue, let me know...
[demo3.zip](https://github.com/realthunder/fcad_pcb/files/4589642/demo3.zip)

realthunder, 2020-05-06:
Fixed. Please check.

closed 2020-05-06 by realthunder

9robs, 2020-05-07:
Works like a charm - much appreciated...where should I send the "thank you" beers

realthunder, 2020-05-07:
You can click any donation badge [here](https://github.com/realthunder/FreeCAD_assembly3#assembly3-workbench-for-freecad). Thanks!

Maybe you will also be interested to follow my FreeCAD development there.

commit 8bbca129 (2020-05-06): Fix layer filtering when making pads

Fixes #25