realthunder/fcad_pcb #27: pcbnew nightly issue

easyw, 2020-11-20:
Recently in kicad 5.99 pcbnew nightly format the 'module' has been changed to 'footprint'
The best would be to keep compatibility for kicad v5 and v6 (5.99 nightly)

kicad v5.1.x
```
  (module to220-5-horizontal-inverted (layer F.Cu) (tedit 5E344340) (tstamp 5FB81C5D)
    (at 116.84 114.3)
    (descr "TO220HInv, 5pins")

```
kicad v5.99 (nightly)
```
  (footprint "to220-5-horizontal-inverted" (layer "F.Cu")
    (tedit 5E344340) (tstamp 00000000-0000-0000-0000-00005fb81c5d)
    (at 116.84 114.3)
    (descr "TO220HInv, 5pins")

```
I'm attaching a sample board in kicad v5 and v6
[test-v6.kicad_pcb.zip](https://github.com/realthunder/fcad_pcb/files/5576139/test-v6.kicad_pcb.zip)
[test-v5.kicad_pcb.zip](https://github.com/realthunder/fcad_pcb/files/5576140/test-v5.kicad_pcb.zip)

realthunder, 2020-11-30:
Fixed. Please recursively sync the submodules as well.

closed 2020-11-29 by realthunder

easyw, 2020-11-30:
@ realthunder just a minor issue during my testing...
please add
`               'fp_arc',` after this line
[kicad_pcb.py#L52](https://github.com/realthunder/kicad_parser/blob/a5a1e71bba2076de2f0d7153895a4fccc1fbe9d6/kicad_pcb.py#L52)
thanks again for your fast support

commit b67059d8 (2020-11-29): Support new kicad_pcb keyword 'footprint' as alias to 'module'

Fixes #27