realthunder/fcad_pcb #30: Clearence between copper zones distorted

der-ule, 2021-01-28:
Hello @realthunder 

I have been using your macro indirectly through the KiCAD StepUp Workbench, great stuff, thanks a bunch for it! 

However, on my latest board I did notice that some zones where fused together and the space between them was not accurate, I raised the issue back at the [KiCAD Forums ](https://forum.kicad.info/t/kicad-stepup-clearence-between-zones-deformed/27244/4) there @easyw recommended me to raise an issue here to see if you could help.

I followed your installation instruction and loaded my board in FreeCAD using directly your plugin and the problem seems to be still there.

KiCAD
[image]

fcad_pcb
[image]

The output of the macro looks pretty normal to me:
```
09:17:00  A3 available
09:19:01  making copper layer F.Cu...
09:19:01    making pads...
09:19:06      modules: 82
09:19:06      pads: 326, skipped: 41
09:19:06      vias: 117, skipped: 0
09:19:06      total pads added: 402
09:19:28    pads done
09:19:28    making tracks...
09:19:28      making 399 tracks  of width 0.20, (0/535)
09:19:32      making 136 tracks  of width 0.60, (399/535)
09:19:38    tracks done
09:19:38    making zones...
09:19:38      making zone /V_SW...
09:19:38        region 1/1, holes: 0
09:19:38      making zone GND...
09:19:38        region 1/2, holes: 0
09:19:38        region 2/2, holes: 0
09:19:38      making zone GND...
09:19:38        region 1/1, holes: 0
09:19:38      making zone /V_SW...
09:19:38        region 1/1, holes: 0
09:19:38      making zone +3V3...
09:19:38        region 1/1, holes: 31
09:19:39      making zone /V_SW...
09:19:39        region 1/1, holes: 0
09:19:39      making zone "Net-(C503-Pad2)"...
09:19:39        region 1/1, holes: 0
09:19:39      making zone /VBAT+...
09:19:39        region 1/1, holes: 0
09:19:39      making zone ""...
09:19:39      making zone "Net-(C206-Pad2)"...
09:19:39        region 1/1, holes: 0
09:19:39      making zone "Net-(L203-Pad1)"...
09:19:39        region 1/1, holes: 0
09:19:39      making zone "Net-(C207-Pad2)"...
09:19:39        region 1/1, holes: 0
09:19:39      making zone "Net-(L204-Pad1)"...
09:19:39        region 1/1, holes: 0
09:19:39      making zone GND...
09:19:39        region 1/1, holes: 0
09:19:40      making zone /V_SW...
09:19:40        region 1/1, holes: 0
09:19:40      making zone "Net-(C205-Pad2)"...
09:19:40        region 1/1, holes: 0
09:19:40      making zone "Net-(C208-Pad2)"...
09:19:40        region 1/1, holes: 4
09:19:40      making zone /V_SW...
09:19:40        region 1/1, holes: 0
09:19:40      making zone "Net-(L201-Pad1)"...
09:19:40        region 1/1, holes: 0
09:19:40      making zone GND...
09:19:40        region 1/1, holes: 0
09:19:40      making zone "Net-(L202-Pad1)"...
09:19:40        region 1/1, holes: 0
09:19:40      making zone "Net-(C205-Pad2)"...
09:19:40        region 1/1, holes: 4
09:19:40      making zone /V_USB...
09:19:40        region 1/1, holes: 0
09:19:40    zones done
09:19:45  done copper layer F.Cu
```
My FreeCAD version is:
```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23323 (Git)
Build type: Release
Branch: master
Hash: 512d5c6141aec52b6eecc67370336a28fde862a6
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: German/Germany (de_DE)
```
Any my KiCAD version is:
```
Application: Pcbnew
Version: (5.1.8)-1, release build
Libraries:
    wxWidgets 3.0.5
    libcurl/7.71.0 OpenSSL/1.1.1g (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.41.0
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.73.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.71.0
    Compiler: GCC 10.2.0 with C++ ABI 1014

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

Let me know if I can somehow assist you further and once again, thank you for your time and effort!

P.S. I haven't been able to reproduce the problem with a test board, as such here is the original board with the problem: 
[charger.zip](https://github.com/realthunder/fcad_pcb/files/5885773/charger.zip)

realthunder, 2021-01-29:
Can you please sync my repo and check if the problem is fixed?

der-ule, 2021-01-29:
Wow! It worked flawlessly with the latest version!

[image]

Thanks a lot once again for this magnificent software and your time and dedication!

```
08:06:01  A3 available
08:10:36  making pcb...
08:10:36    making board...
08:10:36      making 23 lines
08:10:36      making 4 arcs
08:10:36      making 2 polys
08:10:37        making holes...
08:10:37          pad holes: 25, skipped: 0
08:10:37          oval holes: 0
08:10:37          via holes: 117, skipped: 0
08:10:37          total holes added: 142
08:10:37        holes done
08:10:38    board done
08:10:38    making all copper layers...
08:10:38      fetch holes 'holes_wire' from cache
08:10:38      making copper layer F.Cu...
08:10:38        making pads...
08:10:43          modules: 82
08:10:43          pads: 326, skipped: 41
08:10:43          vias: 117, skipped: 0
08:10:43          total pads added: 402
08:10:59        pads done
08:10:59        making tracks...
08:10:59          making 399 tracks  of width 0.20, (0/535)
08:11:03          making 136 tracks  of width 0.60, (399/535)
08:11:07        tracks done
08:11:07        making zones...
08:11:07          making zone /V_SW...
08:11:07            region 1/1, holes: 0
08:11:07          making zone GND...
08:11:07            region 1/2, holes: 0
08:11:07            region 2/2, holes: 0
08:11:07          making zone GND...
08:11:07            region 1/1, holes: 0
08:11:07          making zone /V_SW...
08:11:07            region 1/1, holes: 0
08:11:07          making zone +3V3...
08:11:07            region 1/1, holes: 31
08:11:08          making zone /V_SW...
08:11:08            region 1/1, holes: 0
08:11:08          making zone "Net-(C503-Pad2)"...
08:11:08            region 1/1, holes: 0
08:11:08          making zone /VBAT+...
08:11:08            region 1/1, holes: 0
08:11:08          making zone ""...
08:11:08          making zone "Net-(C206-Pad2)"...
08:11:08            region 1/1, holes: 0
08:11:08          making zone "Net-(L203-Pad1)"...
08:11:08            region 1/1, holes: 0
08:11:08          making zone "Net-(C207-Pad2)"...
08:11:08            region 1/1, holes: 0
08:11:08          making zone "Net-(L204-Pad1)"...
08:11:08            region 1/1, holes: 0
08:11:08          making zone GND...
08:11:08            region 1/1, holes: 0
08:11:08          making zone /V_SW...
08:11:08            region 1/1, holes: 0
08:11:08          making zone "Net-(C205-Pad2)"...
08:11:09            region 1/1, holes: 0
08:11:09          making zone "Net-(C208-Pad2)"...
08:11:09            region 1/1, holes: 4
08:11:09          making zone /V_SW...
08:11:09            region 1/1, holes: 0
[... 72 lines cut ...]
08:15:03      making zone GND...
08:15:03        region 1/1, holes: 0
08:15:03      making zone /V_SW...
08:15:03        region 1/1, holes: 0
08:15:03      making zone "Net-(C205-Pad2)"...
08:15:03        region 1/1, holes: 0
08:15:03      making zone "Net-(C208-Pad2)"...
08:15:03        region 1/1, holes: 4
08:15:03      making zone /V_SW...
08:15:03        region 1/1, holes: 0
08:15:03      making zone "Net-(L201-Pad1)"...
08:15:03        region 1/1, holes: 0
08:15:03      making zone GND...
08:15:03        region 1/1, holes: 0
08:15:03      making zone "Net-(L202-Pad1)"...
08:15:03        region 1/1, holes: 0
08:15:03      making zone "Net-(C205-Pad2)"...
08:15:03        region 1/1, holes: 4
08:15:04      making zone /V_USB...
08:15:04        region 1/1, holes: 0
08:15:04    zones done
08:15:13  done copper layer F.Cu
```

p.s. wouldn't it be good idea to output the software version somewhere at the top when loading the plugin ? I wasn't sure if freecad picked  up the changes until I saw that there was no error anymore.

luzpaz, 2022-10-30:
> p.s. wouldn't it be good idea to output the software version somewhere at the top when loading the plugin ? I wasn't sure if freecad picked up the changes until I saw that there was no error anymore.

@realthunder do you want a new ticket for this?