realthunder/fcad_pcb #36: macro not working on MacOS

veryLazyTom, 2021-04-24:
Hi real thunder,

I have tried to run your macro on FreeCAD 0.20 on MacOS but was not able.

The macro is installed in the following directory:

/Users/marco/Library/Preferences/FreeCAD/Macro/fcad_pcb

what I do:
I set the working directory to /Users/marco/Library/Preferences/FreeCAD/Macro/fcad_pcb

and then try to run it in the console by writing

from fcad_pcb import kicad

and the result I get is the following:

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/marco/Library/Preferences/FreeCAD/Macro/fcad_pcb/kicad.py", line 4, in <module>
    from future.utils import iteritems
ModuleNotFoundError: No module named 'future'

any suggestion on how to address the problem?

Thanks,

Marco

realthunder, 2021-04-24:
Please install `future` module through pip3.

```
pip3 install future
```

veryLazyTom, 2021-04-24:
thanks for the fast reply, 

I have limited experience with git/related software.

after saying: pip3 install future

I've got this output:

-bash: pip3: command not found

what is pip3?

what do I need to install to have pip3 working?

realthunder, 2021-04-24:
Do you have [homebrew](https://brew.sh/) installed on your Mac? If not, just run the command in the linked webpage to install it. And then install Python3,

```
brew install python3
```

After that you should have pip3 available in `/usr/local/bin` directory. By right, this path should already be in your search path list, so you can simply type `pip3 install future`.

veryLazyTom, 2021-04-25:
Hi, homebrew might be an issue on my mac so I'll keep this on hold and open another issue which is more important now.

veryLazyTom, 2021-04-27:
just you are aware, I had the exact same issue on the PC, even after installing python3.

In any case, I just updated the repo as explained in https://github.com/realthunder/fcad_pcb/issues/37 and everything works ok on both Mac and PC.

Thanks!

closed 2021-04-27 by veryLazyTom