realthunder/fcad_pcb #37: trace not modelled properly while importing to FC

veryLazyTom, 2021-04-25:
Hi,

although I'm having problems using an updated version of your scripts (see https://github.com/realthunder/fcad_pcb/issues/36) I have tried to run an older working version of your scripts that I have on my laptop and although the script works correctly (checked with your test files) I run into a problem.

You can see the expected trace shown in figure trace_designed_Kicad.jpeg
[image]

But when I try to import it in FreeCAD and say (in Freecad):
from fcad_pcb import kicad
pcb = kicad.KicadFcad(<FILE_PATH_AND_FILENAME>)
pcb.make(copper_thickness=0.035, board_thickness=12, combo=False, fuseCoppers=True )
I get the result shown in figure trace_problem_during_import_FC.jpeg
[image]

As you can see the traces are all fused in a single copper area. what can be the problem?

Please note that the winding is a pour copper area.

I send you the Kicad project so that you can have all the details of the design.

I hope you can help me with this problem,

Thanks,

Marco

[FC_trafo_V01.zip](https://github.com/realthunder/fcad_pcb/files/6371696/FC_trafo_V01.zip)

realthunder, 2021-04-27:
I just tried this file on my PC, and it works as expected. The problem you experience here may have been fixed along the way, as you mention that you are using an older version of this script. I wonder what problem prevent you from using the latest version. It can't be the `future` module, because it is there from the beginning. Anyway, I have just removed use of `future` module. Maybe you can try again with the latest version.

veryLazyTom, 2021-04-27:
just updated the repo and it works on both PC and Mac ^^

to be precise on PC it runs on a stable 0.19 and on Mac on development 0.20 

fantastic piece of work, thanks man!

closed 2021-04-27 by veryLazyTom