realthunder/fcad_pcb #38: inconsistent shape of via after applying  pcb.via_bound=1

veryLazyTom, 2021-04-27:
Hi, 

after reading https://github.com/realthunder/fcad_pcb/issues/22 I tried the feature via_bound=1 on my model and I've noticed that although all via were filled, not all not all of them were changed to rectangular.

see the picture below
[image]

Can you please have a look into this? I send you the Kicad file as well

[test_V01.zip](https://github.com/realthunder/fcad_pcb/files/6384523/test_V01.zip)

As a separate request, when via_bound=1 is used, would it be possible to have a choice on wether you want a circular or square via?

Thanks, 

Marco

veryLazyTom, 2021-05-01:
Hi,

I see. the main reason for me to rise an issue was that it looked like an inconsistency.
So if the outcome is as expected, I would not ask to add this feature.

with regards to keeping vias as they are but without holes, I have tried the command

`pcb = kicad.KicadFcad('path+name', via_bound=0, via_skip_hole=True)
pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True)`

but it didn't work. What command should I use? 

Just you are aware, the goal is to send the complete PCB (copper + dielectric layers) to a mesher.

Also, I have a couple of requests that I would like to ask, mainly to simplify the handling of the PCB when imported in a mesher. Is it possible?

realthunder, 2021-05-05:
I just pushed a commit to allow `via_skip_hole` without `via_bound`. You can also make coppers without any hole at all by running `pcb.makeCoppers('solid', fuse=True)`. The command to make board is simply `pcb.makeBoard()`.

Regarding the PCB meshing, sure, just ask. I'll see what I can do.

veryLazyTom, 2021-05-08:
Awesome!

with regard to the term `pushed a commit`, sorry but I'm not yet familiar with git terminology, does this mean that the feature or it is already implemented?
I'm asking this because when I checked the git status of my local repository (installed before your commit ) I got the answer that it is all updated and there is no need to do anything. This should not be the case if you have already updated the code.

About the requests:

1. When filtering multiple nets (for example: `pcb.setNetFilter('GND','VCC')`), is it possible to have a solid body for each net?

2. Can `pcb.setNetFilter` include in its filter list the bare PCB (I mean, only the FR4) ?

3. When building copper traces and/or bare PCB, is it possible to include in the macro a list of dielectric layer thicknesses (as done for Cu layers)? I'd like to make an observation on this point. I don't know if you are already aware of this but just in case, Kicad V5.1.10 (current stable release) PCB file does not have good dielectric thickness info however V5.99 (and I assume the upcoming stable V6.00 will do the same) allows to specify dielectric thickness for each layer so from the next release it should be possible to extract this info directly form the PCB file.

4. Once copper trace bodies are created, Is it possible to have a list of the Z coordinates (with respect to global coordinates) of the top and bottom surface of each copper layer?

5. When working with 2D objects (that is, using `pcb.setLayer()` and `pcb.makeCopper()`), is it possible to have through holes and vias built as described in the picture below?

[image]

6. For full 3D copper bodies, is it possible to have through holes and vias  build as described in figure 2 above? 
   
This should be it :)

Marco

veryLazyTom, 2021-06-16:
Hi,

it has been some time since I posted my requests. Did you get any chance to have a look or work on them?

Let me know if you have any question or there is any problem with them.

Marco

realthunder, 2021-06-17:
Somehow you post has slipped my mind. Sorry for not responding.

> with regard to the term `pushed a commit`, sorry but I'm not yet familiar with git terminology, does this mean that the feature is already implemented?

You need to pull the change from remote to get the update. `git status` only shows the local changes. If you are using command line tools, then run inside the repo directory `git pull`. If you are using some kind of desktop client, try to find the command with the `pull` keyword.

> 1. When filtering multiple nets (for example: `pcb.setNetFilter('GND','VCC')`), is it possible to have a solid body for each net?

Yes you can. After set netfilter you can call `pcb.makeCoppers()` as usual to get the solid. Note that I just pushed a commit to fix the missing filter reset when calling `setNetFilter()`. Just run `git pull` to get the update.

> 2. Can `pcb.setNetFilter` include in its filter list the bare PCB (I mean, only the FR4) ?

The layer is filtered by call to `self.setLayer()`. But I am not sure what you do you want to extract from the FR4 layer?

> 3. When building copper traces and/or bare PCB, is it possible to include in the macro a list of dielectric layer thicknesses (as done for Cu layers)?

Can you provide a file containing the thickness info? If you just want to print out the thickness info, it can be done without modification. I just need the file to find out the parameter path.

> 4\. Once copper trace bodies are created, Is it possible to have a list of the Z coordinates (with respect to global coordinates) of the top and bottom surface of each copper layer?

Assuming you created a fused copper layer object with the following command
```python
coppers = pcb.makeCoppers(shape_type='solid',holes=True,fuse=True,thickness=0.05)
```

You can use the following command to print out the z top and bottom for each layer
```python
for obj in coppers.Base.Shapes:
  bbox = obj.Shape.BoundBox
  print('%s: %s, %s' % (obj.Label, bbox.ZMin, bbox.ZMax))
```

veryLazyTom, 2021-06-19:
Hi realthunder, 

thanks for the reply, I'll try out the commands and give you the file with thickness asap.

About your question on why do I need the FR4, there are couple of reasons: you need it to calculate capacitance between nets and it can be helpful for creating additional partitioning to help optimizing the mesh.