realthunder/fcad_pcb #42: missing feature: Net Tie as tracks

easyw, 2021-09-04:
In some RF user case, it is useful to model a footprint partially as Pad ans partially as Net Tie. This will allow to solve DRC issues.

It would be nice to load also *.Cu Net Tie graphic as tacks.

Here a minimal sample file:
[NetTie-minimal-test.zip](https://github.com/realthunder/fcad_pcb/files/7109467/NetTie-minimal-test.zip)
the kicad attribute is `fp_poly` on Cu layers.

closed 2021-09-24 by realthunder

easyw, 2021-09-26:
@realthunder 
thanks for your support.

'NetTie-minimal-test.zip' works with your commit https://github.com/realthunder/fcad_pcb/commit/14e9bf5537ef3f13503ec83647a8d689d8ce3b02
but it seems this one https://github.com/realthunder/fcad_pcb/commit/0b1996598b1a1619c9d4fec0e1bc4259c9ecb82a is breaking the functionality
`11:53:43  Traceback (most recent call last):
  File "C:/Cad/Prj_K/3D-FreeCad-tools/fcad_pcb-test-3.py", line 73, in <module>
    pcb.make(copper_thickness=0.035, load_parts=True, board_thickness=1.53, combo=False, fuseCoppers=True )    
  File "d:fcad_pcb\kicad.py", line 2183, in make
    objs.append(self.makeBoard(prefix=None,thickness=board_thickness))
  File "d:fcad_pcb\kicad.py", line 1256, in makeBoard
    thickness = layers[0][1]
<class 'TypeError'>: 'float' object is not subscriptable`

Moreover 'NetTie-minimal-test-2.zip' in my tests is not working also with the previous patch
`12:06:58  Traceback (most recent call last):
  File "C:/Cad/Prj_K/3D-FreeCad-tools/fcad_pcb-test-3.py", line 73, in <module>
    pcb.make(copper_thickness=0.035, load_parts=True, board_thickness=1.53, combo=False, fuseCoppers=True )    
  File "d:fcad_pcb\kicad.py", line 2040, in make
    coppers = self.makeCoppers(shape_type='solid',holes=True,prefix=None,
  File "d:fcad_pcb\kicad.py", line 1875, in makeCoppers
    copper = self.makeCopper(shape_type,t,fit_arcs=fit_arcs,
  File "d:fcad_pcb\kicad.py", line 1793, in makeCopper
    obj = getattr(self,'make{}'.format(name))(fit_arcs=sub_fit_arcs,
  File "d:fcad_pcb\kicad.py", line 1481, in makePads
    self._makeShape(m, 'fp', pads)
  File "d:fcad_pcb\kicad.py", line 1072, in _makeShape
    wires.append(self._makeWires(
  File "d:fcad_pcb\kicad.py", line 883, in _makeWires
    return self._makeArea(obj,name,offset=offset,fill=fill,
  File "d:fcad_pcb\kicad.py", line 855, in _makeArea
    ret = ret.getShape()
<class 'ValueError'>: {'sclassname': 'class Base::ValueError', 'sErrMsg': 'shapes are not planar', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'shapes are not planar', 'btranslatable': False, 'breported': False}
`

realthunder, 2021-09-26:
Fixed. Please try.

realthunder, 2021-09-26:
BTW, I have added support for board stackup. So, if you do not specify `board_thickness` when calling `make()` for your sample file you'll find the board consisting of three dielectric layers.

easyw, 2021-09-26:
> Fixed. Please try.

thanks a lot...
I 'm noticing two small issues:
1) when specifying a  board_thickness it seems the tracks go to the wrong layer: bot instead of top
2) with 'NetTie-minimal-test-2.zip' I get a wrong width for the fp_line track (3mm instead of 1mm)

kicadstepup, 2021-09-27:
@realthunder 
many thanks!

easyw, 2021-09-27:
@realthunder 
I'm still getting a small issue...
with the attached board, everything is fine if I give the `board_thickness`, but there is an issue if I don't specify this parameter
[coupler-rf.zip](https://github.com/realthunder/fcad_pcb/files/7236044/coupler-rf.zip)

[image]

commit 14e9bf55 (2021-09-24): Support fp shapes in pad

Fixes #42
@easyw