realthunder/fcad_pcb #50: no module named fcad_pcb

robotfishe, 2021-11-12:
This might be a really simple problem, but I've been unable to fix it on two Windows PCs and one Ubuntu one. I follow the instructions to clone the repositories and check out the submodules into my macros directory, so I have an fcad_pcb directory inside my macro directory with kicad.py, kicad_parser, etc inside that.

When I try entering "from fcad_pcb import kicad" in the python console in FreeCAD, I get the following error:

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\FreeCAD\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'fcad_pcb'

Any ideas?

realthunder, 2021-11-12:
Maybe you didn't get the correct Macro directory?

[image]

robotfishe, 2021-11-12:
Nope, it's definitely not that:

[image]
[image]
[image]

realthunder, 2021-11-12:
Ah, I see it. You miss spell the module name. Should be `fcad_pcb`. `fcad` is short for `FreeCAD`.

robotfishe, 2021-11-12:
Oops, made a typo that time, but I get the same error when it's spelled correctly.

[image]

realthunder, 2021-11-12:
That's strange. Have you ever tried other macros? Do they work? Can you find your macro directory in `sys.path`?

```
import sys
sys.path
```

robotfishe, 2021-11-13:
Running sys.path was helpful in a weird way! The macros directory under AppData was there, but I also noticed that there was another Macros path listed, C:\\Program Files\\FreeCAD\\Macros. I copied fcad_pcb into there, restarted FreeCAD, and suddenly I'm able to import.

Still no idea why it was failing to pick up from the correct place, though.

realthunder, 2021-11-13:
Maybe some permission issue.

closed 2021-11-13 by robotfishe