realthunder/fcad_pcb #55: Cannot install EasyAlias Realthunder Windows 11

Badgersurfer, 2022-02-22:
I want to get into spreadsheets but I watched a YT video that said I need to install the macro EasyAlias. I followed his procedure but is says it cannot install it. Also I have no macros showing similar to his. I need some help here please as I am obviously missing something..

realthunder, 2022-02-24:
This does not seem to be related to fcad_pcb repo. But anyway, are there any error output?

Badgersurfer, 2022-02-25:
No errors I just go to the Addons Manager, click on the Macros section, select the EasyAlias macro and hit the install button and it says cannot install. I have also tried it in the standard FreeCad 19.3 package and it is just the same so I have no idea what is wrong. I thought you may have an idea? I cannot use spreadsheets without it. I was watching a guy on YT using the spreadsheet workbench along with the EasyAlias macro it looked easy to do so that is when I tried to install the macro. But alas it won't install. 

[image]
[image]

realthunder, 2022-02-25:
I just checked source code of AddonManager. It looks like your Macro directory is not set, or is not writable. Make sure to select a writable directory in `Macro -> Macros...`

[image]

Badgersurfer, 2022-02-25:
I am just checking it now. Do you mean this one... My home directory is c:/Freecad Realthunder... I just created a dir called home/but it doesn't allow me to point to it. 
I have attached where the macro folder is installed by default. Not sure what I need to do

[image]
?

Badgersurfer, 2022-02-25:
Sorry I was being a bit thick there... yep I get what you mean...
[image]
.

Badgersurfer, 2022-02-25:
I have done this, restarted Freecad and then gone back to addons manager, selected Macros>EasyAlias>Install but is still says cannot install. Did you manage to install it?

Badgersurfer, 2022-02-25:
I have done it. When you use the browse button to locate the folder to change it to one of your own choice (like you suggested) it won't pick up the folder, it asks for a filename and then selects the filename and adds it to the Macros path. I manually removed the filename and restarted and it works fine. Thank you for your help. I will make a video of that now for future reference and in case anyone else has the dsame issue.

Badgersurfer, 2022-02-25:
[image]