realthunder/fcad_pcb #58: bottom tracks issue (kv6)

easyw, 2022-07-14:
hi, I have an issue in loading B.Cu tracks with the attached board
[stackup-issue-saved-test.zip](https://github.com/realthunder/fcad_pcb/files/9110206/stackup-issue-saved-test.zip)
[image]

realthunder, 2022-07-16:
What is you command to make the track?

Also, can you please tell why layer `dieletric 1` has two thickness value. What does the `addsublayer` thing do?

```
(layer "dielectric 1" (type "prepreg") (thickness 0.037617 locked) (material "106 185HR") (epsilon_r 3.74) (loss_tangent 0.02)
        addsublayer (thickness 0.087757 locked) (material "2113 185HR") (epsilon_r 4.02) (loss_tangent 0))
```

easyw, 2022-07-17:
> What is you command to make the track?

[code]
import fcad_pcb
from fcad_pcb import kicad
from kicad_parser import KicadPCB
pcb = kicad.KicadFcad(filename)
pcb.setLayer('B.Cu')
pcb.makeZones(shape_type='face',thickness=0.05, fit_arcs=True,holes=True)
pcb.makeTracks(holes=True)
pcb.makePads(shape_type='face',thickness=0.05,holes=True,fit_arcs=True)
[/code]
or
[code]
import fcad_pcb
from fcad_pcb import kicad
from kicad_parser import KicadPCB
pcb = kicad.KicadFcad(filename)
pcb.setLayer('B.Cu')
coppers = pcb.makeCoppers(shape_type='solid', holes=True, fuse=True)
[/code]

> Also, can you please tell why layer `dieletric 1` has two thickness value. What does the `addsublayer` thing do?

that was a setting from a user, which replicated a failing user case in this pcb

easyw, 2022-07-18:
@realthunder 
I forgot to add the FC version
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.21.29484 (Git)
Build type: Release
Branch: master
Hash: 4de0e6fb198a60bd09400d64994e7aa8308b1e23
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
[/code]

realthunder, 2022-07-21:
The shape making failed because the tricky hole cutting shown below. You can fix it by increasing the arc fitting accuracy like below. The default accuracy (actually tolerance) is 5e-4.

```python
pcb = kicad.KicadFcad(filename, arc_fit_accuracy=1e-4)
```

[image]

easyw, 2022-07-21:
@realthunder 
I tried but still in some user case I cannot solve the issue...
then I tried:
1) create tracks without holes
2) create holes
3) extrude holes as solid
4) make a cut between track faces and solid extrude holes
and I got in some case an issue in the result...
then I tried to change step 4) with a 'fuzzy' cut (available since OCC 7)
4) make a 'fuzzy' cut between track faces and solid extrude holes
and this is giving me, ATM, always a correct result
https://dev.opencascade.org/doc/occt-7.3.0/overview/html/occt_user_guides__boolean_operations.html#occt_algorithms_11a_1

Do you think it could be possible to add a cut with fuzzy option in fcad_pcb?

> The following results are obtained using Basic Operations and the Fuzzy ones with the fuzzy value 6e-5

realthunder, 2022-07-26:
Can you please collect some files where the cut failed? I'd like to dig a bit deeper into those failure cases.

easyw, 2022-07-28:
@realthunder 
I'm sorry but I cannot replicate the issue on other boards atm ... I need to dig deeply in my testing board folder...