realthunder/fcad_pcb #59: Trouble with minimum width parameter of polygons

TheRealTobiwan, 2022-07-20:
Hi,
I am using your code to extract the copper traces from my pcb's for FEM simulations and so far this works well. But I think I recently found an issue while I was working on a comparatively small pcb. I extracted the copper using:

`from fcad_pcb import kicad

pcb = kicad.KicadFcad(<full_path_to_your_kicad_pcb_file>)

pcb.make(copper_thickness=0.035, board_thickness=1.53, combo=False, fuseCoppers=True )`

but it seems like all my polygons are a little to big. Here is my example pcb in Kicad. The distance between the highlighted polygons is 0.2mm:
[image]
And here is what I get in FreeCAD using the commands above:
[image]
As you can see the distance is 0.
I tried some things and I found that the "Minimum Width" Parameter in the polygons settings seems to impact this behaviour (see next picture for the setting). 
[image]
For the test above it was set to 0.2mm. If I reduce it to 0.0254mm for both polygons I get the following result:
[image]
The change of "Minimum Width" from 0.2mm to 0.0254mm (which seems to be the minimum in Kicad) meens a difference of approx. 0.17mm, which seems to correlate with the new distance (see picture above).
For me it seems like the parameter "minimum width" is interpreted as some kind of "line thickness" of the polygons outline, which is than "added" to the original polygon.

My Toolchain:
Kicad 6.06
FreeCAD 0.20
latest version of your code

Last but not least: Big thanks for your effort, your code is really useful!

realthunder, 2022-07-20:
There is an interesting [read](https://bugs.launchpad.net/kicad/+bug/1658660) about this `minimum width` issue. Maybe they finally decide to change the behavior some how. But looks like it will break backward compatibility. Can you by any chance try your file in older version KiCAD? Also please confirm it is not just a display thing. I mean just to be sure, check the KiCAD 3D view and see the zone shape. Maybe you can also send me a simplified pcb showing this problem.

easyw, 2022-07-20:
@TheRealTobiwan 
thanks for your info... I was hoping to get some tips to do FEM in FC

realthunder, 2022-07-20:
Yes, please send me a sample file. I am pretty sure KiCAD used to use `Minimum width` to inflate the zone for reasons discussed in the [thread](https://bugs.launchpad.net/kicad/+bug/1658660) I quoted last time. I even have a code [comment](https://github.com/realthunder/fcad_pcb/blob/8f850f1c38562aef40fd4a6f0d0fed6c1e1b6eef/kicad.py#L1856) for this quite confusing behavior. We need to find out when did KiCAD changed meaning of this option, and if there is any other attribute ensuring backward compatibility.

BTW, as a temporary fix, you can pass a `zone_inflate` parameter when creating pcb to counter the `minimum width` inflation, like below assuming your `minimum width` setting is 0.2,

```python
pcb = kicad.KicadFcad(<full_path_to_your_kicad_pcb_file>, zone_inflate=-0.1)
```
You can change zone_inflate any time. It basically let you manually shrink or expand the zone.

```python
pcb.zone_inflate = -0.2
```

juddidier, 2022-10-17:
Hi realthunder,

I'm joining the "party" because I have the same problem and like to share my findings:
in <kicad.py> line 1869 you add a "thickness*0.5" to the offset (in the version before it was "z.min_thickness"). IMHO this is erroneous and can be deleted.
Eg: In KiCAD my copper zone aligns exactly with the pad (what I wand):
[image]

When importing the pcb in FreeCAD the copper zone is slightly to big (exactly 0.0127mm or "thickness*0.5").
[image]

from what I read in your older version of <kicad.py>, I think there was a misunderstanding of what the purpose of "minimum with" (which fills the variable z.min_thickness) in the copper zone properties is.

If you have compatibility concerns, couldn't you use the version date saved in the first line of *.kicad_pcb to make an "if..else" decision: compatibility mode or not?