realthunder/fcad_pcb #66: Zone not imported if zone active for multiple layers

aa956, 2023-04-11:
Zone will not be imported if zone is active for both F.Cu and B.Cu layers.
Probably other combination of layers (F, B, In.1, In.2 etc will be affected too):

[image]

Zones are successfully imported if there are 2 zones (duplicates of each other), one of them only in F.Cu, other only in B.Cu

realthunder, 2024-02-03:
Could you please provide a sample file for this?

aa956, 2024-02-03:
Attached the KiCAD example project.

Recreated with freshly reinstalled KiCAD 7.0.10, tried to import in fresh FreeCAD 0.21.2 install with newly checked out master branch of fcad_pcb (currently f413e02c36fbcd73165992b800e235ac49117228):

Still the same issue, looks like PCB multi layer zone (`(layers "F&B.Cu")` in the attached project) is problematic:

```
(zone (net 2) (net_name "GND") (layers "F&B.Cu") (tstamp 25dbf940-302e-4dc3-8b7b-d484cd6cc2da) (name "global_gnd") (hatch edge 0.5)
```
[copper_area_issue66.zip](https://github.com/realthunder/fcad_pcb/files/14154231/copper_area_issue66.zip)

commit d8d0b060 (2024-02-04): Fix quoted layer filtering

Fixes #66
@aa956
closed 2024-02-04 by realthunder