realthunder/fcad_pcb #67: Castellated holes import

ONLYA, 2023-04-14:
I created castellated holes by putting them at the edge of the PCB. When I import the PCB by `pcb.make(...)`, the entire holes are imported at the edge instead of semi-cut holes as shown in the KiCad's internal 3D viewer.
[image]

Is there an already made feature that I do not know or it is still awaiting to be implemented?

realthunder, 2024-02-03:
Could you please provide a sample file with this feature?

realthunder, 2024-02-11:
Just added this feature. It's experimental at the moment. You're welcome to test it. The turn it on, pass in `castellated=True` when load your pcb file, as shown below,

```
pcb = kicad.KicadFcad('/path/to/some.kicad_pcb', castellated=True)
```
After which any of the `pcb.makeXXX()` functions shall produce castellated coppers.