realthunder/fcad_pcb #68: Vias on wrong layers

qaqcvc, 2023-06-24:
First of all, I would like to thank all the contributors for this wonderful tool. It is a lifesaver!

In this practive, I am tring to output the pcb with micro layers from 1-2 and 23-24, as well as burried vias from 2-23. But the output shows wrong vias locations:
[image]

Is it a bug, or do I possibly set smethong wrong?
I have checked the setting of vias in Kicad, they are all rightly set with right 3d view in Kicad.
[image]

With a deeper observation, it seems that in Freecad, all vias more moved 2 layers below. But the holes are rightly located.
[image]

qaqcvc, 2023-06-24:
I am using kicad 6 and 7 (test both), and Freecad 0.19

realthunder, 2023-06-27:
Is it possible to post a sample file showing the problem?

qaqcvc, 2023-06-28:
Hi realthunder, thank you for checking! Please find one minimum test file here: [test.zip](https://github.com/realthunder/fcad_pcb/files/11899503/test.zip)

There are several issues in this practice:
1. I need to at least add one through via, otherwise it gives me error:  line 991, in _makeArea
    if not force and obj[0].TypeId == 'Path::FeatureArea' and (
<class 'IndexError'>: list index out of range
2. Normally when I select add via annular ring only for connected layers, all rings will be generated.
3. 1-2 micro vias and middle vias are on wrong positions
[image]
4. Wrong holes on top and bottom layers
[image]
[image]
5. Although successfully generate the 3d model but got error, this error occasionally occurs.
line 11, in <module>     Part.show(coppers)   <class 'TypeError'>: argument 1 must be Part.Shape, not Part.Feature

Thank you very much realthunder! As a hardware engineer, I tried to use chatgpt to find out possible solutions, but no good. If this issue can be solved, it would be really helpful for folks working on complex multiple layers design!

realthunder, 2023-06-29:
Just added some fixes. Please sync the repo and test. You can close the issue if everything is fine.

qaqcvc, 2023-06-30:
Hi realthunder, thanks for your reponse! From my side, the issue seems to consist:
[image]
[image]

Thank you for your precious time.

realthunder, 2023-06-30:
Have you synced the repo, `git pull` maybe? Is the following model correct?

[image]

[image]

[image]

qaqcvc, 2023-06-30:
Hi realthunder, Sorry I am not familiar with git, do not know why issue occurs when directly download the complete codes. When I only download kicad.py, it works! Now I do not need to seprately generate top,middle and bot layers.

But if the layout does not contain any through via, the fuse cannot be made. This issue can be escaped by mannually adding a dommy through via though.
[image]

And I constantly got this error:
[image]

It would be great if 'connected layers only' is implemented, which can largely reduce mesh complexity by removing unnecessary rings.
[image]
[image]

Thank you for your time!

realthunder, 2023-07-01:
> But if the layout does not contain any through via, the fuse cannot be made. This issue can be escaped by mannually adding a dommy through via though.

Please always provide a sample file showing the problem.

> And I constantly got this error:

The macro code you are using is not mine. Is it hosted in some git repo so that I can take a look?

> It would be great if 'connected layers only' is implemented, which can largely reduce mesh complexity by removing unnecessary rings.

Does that means the via is supposed to only have annular ring in the start and end layers?

qaqcvc, 2023-07-01:
> Please always provide a sample file showing the problem.
In you new added "blind_buried_vias.kicad_pcb" file, if you remove the only though hole via, the fuse will not be made.
[image]

> The macro code you are using is not mine. Is it hosted in some git repo so that I can take a look?
I am using these codes:
```
from fcad_pcb import kicad
pcb = kicad.KicadFcad(r'C:\Users\25423\AppData\Roaming\FreeCAD\Macro\fcad_pcb\tests\blind_buried_vias.kicad_pcb')
pcb.add_feature = True

coppers = pcb.makeCoppers(shape_type='solid',holes=True,fuse=True)
Part.show(coppers)
```

> Does that means the via is supposed to only have annular ring in the start and end layers?
For example, if vias are connected with layer 1,3,5... There should be no annular rings at layer 2,4.., because the fuse doesnot need to connect with these layers. The added annular rings only add unnecessary mesh elements. In my PCB design, there are more than 200 vias. The additional elements could bring addtional several hours of simulation.

realthunder, 2023-07-02:
> In you new added "blind_buried_vias.kicad_pcb" file, if you remove the only though hole via, the fuse will not be made.

This problem has been fixed. Please download kicad.py and try again.

> And I constantly got this error:
> [image]

`Part.show()` is for `Shape` not `Feature`. It is only necessary if `pcb.add_feature = False`, otherwise, there is no need to call that.

> For example, if vias are connected with layer 1,3,5... There should be no annular rings at layer 2,4.., because the fuse doesnot need to connect with these layers.

Can you please make a sample file with this? It seems that KiCad has an option to remove unused vias/pads. Make sure to enable this option.

qaqcvc, 2023-07-02:
Hi realthunder, thnak you! The first two issues are solved.
Regarding the last issue, please ref this file: 
[blind_buried_vias.zip](https://github.com/realthunder/fcad_pcb/files/11931080/blind_buried_vias.zip)
In this layout, there are two though vias, one has annular rings at all layers; while another one has annular rings only with connected layers. 
[image]
[image]

realthunder, 2023-07-03:
I have updated the code. Please try. If you want to be thorough, please make a board with actually connected and unconnected tracks and zones. Maybe you can also try this options on pads (with various orientations), which I haven't implemented yet. I can add that if you can make sample board with those.

qaqcvc, 2023-07-10:
Hi realthunder, sorry for the late reply. I have tested your new kicad.py file, this issue is not solved, I think we have different understanding of the concept of "connected layers". Please have a look on the figure below. 
With connected first layer (connected with trace), the annular ring exists, while for the layers 2,3... , because at these layers this via is not connected to any trace or copper pool, the annular rings need to be removed.
[image]

Here is a sample kicad layout file for you to have a test.
[blind_buried_vias.zip](https://github.com/realthunder/fcad_pcb/files/12005944/blind_buried_vias.zip)

Thank you for your consistent contribution!

qaqcvc, 2023-07-10:
Another question would be,  how can I define the via thickness? Last time I measured in Freecad, seems the thickness of the vias fuse is 35um. Is it possible to modify the thickness by ourselves? Thank you.

realthunder, 2023-07-15:
Please make sure you have updated the kicad.py file. This is what I get when running your files on my compouter.

[image]

> Another question would be, how can I define the via thickness? Last time I measured in Freecad, seems the thickness of the vias fuse is 35um. Is it possible to modify the thickness by ourselves? Thank you.

Do you mean the thickness of the middle cylinder? If it is the annular ring, then the thickness is determined by the layer thickness.

qaqcvc, 2023-07-15:
Hi realthunder, it seems last time I did not update kicad.py successfully.
But with the new kicad.py, I still got one bug with the blid_buried_vias.kicad_pcb file.

I can successfully generate this figure with the original kicad_pcb.
[image]

However, if I press 'B' to fill all zones, then try to generate from Freecad, I will meet this error:
```
11:53:47  making all copper layers...
11:53:47      making holes...
11:53:47        pad holes: 0, skipped: 0
11:53:47        oval holes: 0
11:53:47        total holes added: 8
11:53:47      holes done
11:53:47    making copper layer F.Cu...
11:53:47      making pads...
11:53:47        checking edge cuts
11:53:47          making 10 lines
11:53:47        making 10 lines
11:53:47        checking edge cuts
11:53:47          making 10 lines
11:53:47        making 10 lines
11:53:47  Traceback (most recent call last):
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro/readLayout@001.FCMacro", line 10, in <module>
    coppers = pcb.makeCoppers(shape_type='solid',holes=True,fuse=True)
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 2221, in makeCoppers
    copper = self.makeCopper(shape_type,t,fit_arcs=fit_arcs,
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 2134, in makeCopper
    obj = getattr(self,'make{}'.format(name))(fit_arcs=sub_fit_arcs,
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1815, in makePads
    if filter_unconnected(v, (v.at[0], v.at[1])):
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1723, in filter_unconnected
    excludes = [self.findLayers(s)[0] for s in getattr(v, 'zone_layer_connections', [])]
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1723, in <listcomp>
    excludes = [self.findLayers(s)[0] for s in getattr(v, 'zone_layer_connections', [])]
<class 'AttributeError'>: 'KicadFcad' object has no attribute 'findLayers'
```
You can try to test in your PC with this file, where I just refilled all zones.
[blind_buried_vias.zip](https://github.com/realthunder/fcad_pcb/files/12061225/blind_buried_vias.zip)
If I delete the via with the setting "annular rings connected layers only", layout can be generated successfully.

> Do you mean the thickness of the middle cylinder? If it is the annular ring, then the thickness is determined by the layer thickness.

[image]
Normally the thickness of the covered copper inside vias holes is around 25um. But it would be great if we can have some control of this parameters. For example, one of the common practices is to make the micro vias between 1-2 layers (and 23-24 layers) are filled completelly with copper inside the holes, but for inside blind vias, the covered copper is around 25um.

In your Kicad.py setting, seems it is set defaultlly to 35um?
[image]
It would be perfect to have some flexibiliy on settings of 0.165mm.

Thank you for your precious time!

qaqcvc, 2023-07-28:
Hi realthunder, I have figured out how to set the thickness of the vias. That is by setting the value of hole_size_offset
```
pcb = kicad.KicadFcad(r'file name',hole_size_offset=0.055)
```
By default, the thickness of the vias is 52.5um. And every 0.001 of hole_size_offset represents to reduce the thickness by 0.5um. So 0.055 means 52.5um - 55*0.5um = 25um. I did not go into details to check how this 52.5um is calculated, I just know that several months ago, the default thickness seems to be 35um. Could you briefly explain the method to calculate the thickness?

This error still exists in other test practices.
> However, if I press 'B' to fill all zones, then try to generate from Freecad, I will meet this error:
```
11:53:47  making all copper layers...
11:53:47      making holes...
11:53:47        pad holes: 0, skipped: 0
11:53:47        oval holes: 0
11:53:47        total holes added: 8
11:53:47      holes done
11:53:47    making copper layer F.Cu...
11:53:47      making pads...
11:53:47        checking edge cuts
11:53:47          making 10 lines
11:53:47        making 10 lines
11:53:47        checking edge cuts
11:53:47          making 10 lines
11:53:47        making 10 lines
11:53:47  Traceback (most recent call last):
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro/readLayout@001.FCMacro", line 10, in <module>
    coppers = pcb.makeCoppers(shape_type='solid',holes=True,fuse=True)
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 2221, in makeCoppers
    copper = self.makeCopper(shape_type,t,fit_arcs=fit_arcs,
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 2134, in makeCopper
    obj = getattr(self,'make{}'.format(name))(fit_arcs=sub_fit_arcs,
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1815, in makePads
    if filter_unconnected(v, (v.at[0], v.at[1])):
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1723, in filter_unconnected
    excludes = [self.findLayers(s)[0] for s in getattr(v, 'zone_layer_connections', [])]
  File "C:/Users/25423/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1723, in <listcomp>
    excludes = [self.findLayers(s)[0] for s in getattr(v, 'zone_layer_connections', [])]
<class 'AttributeError'>: 'KicadFcad' object has no attribute 'findLayers'

```

Thank you for your time realthunder.

realthunder, 2023-07-29:
The `findLayers()` is a typo, should be `findLayer()`. It's fixed now.

About the `hole_size_offset`, it specifies the amount of decrease to plated hole diameter. For non-plated holes, this is the amount of increase in diameter. The offset is added to workaround CAD kernel tolerance issue.

realthunder, 2023-08-03:
Could you please post the files that caused those errors?

The base wall thickness of a plated hole is determined by the copper layer thickness setting. Are you using different copper layer thickness? Please provide a file showing variance in hole wall thickness.

qaqcvc, 2023-08-03:
Hi realthunder, you can check with 'blind_buried_vias.kicad_pcb' in your fcad_pcb test folder with the newest kicad.py file.
It will generate this error:
```
  File "D:/Users/t84201115/AppData/Roaming/FreeCAD/Macro/Read_PCBLayout.FCMacro", line 14, in <module>
    coppers = pcb.makeCoppers(shape_type='solid',  fuse=True, holes=True)
  File "D:/Users/t84201115/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 2221, in makeCoppers
    copper = self.makeCopper(shape_type,t,fit_arcs=fit_arcs,
  File "D:/Users/t84201115/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 2134, in makeCopper
    obj = getattr(self,'make{}'.format(name))(fit_arcs=sub_fit_arcs,
  File "D:/Users/t84201115/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1815, in makePads
    if filter_unconnected(v, (v.at[0], v.at[1])):
  File "D:/Users/t84201115/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1723, in filter_unconnected
    excludes = [self.findLayer(s)[0] for s in getattr(v, 'zone_layer_connections', [])]
  File "D:/Users/t84201115/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 1723, in <listcomp>
    excludes = [self.findLayer(s)[0] for s in getattr(v, 'zone_layer_connections', [])]
  File "D:/Users/t84201115/AppData/Roaming/FreeCAD/Macro\fcad_pcb\kicad.py", line 662, in findLayer
    raise KeyError('layer {} not found'.format(layer))
<class 'KeyError'>: 'layer " not found'
```

Right now, I am rolling back to 20220721 kicad.py version.

BTW, if you have time and interest, I found another issue with fcad_pcb to export model to step.
https://github.com/FreeCAD/FreeCAD/issues/10053

Thank you as always.

HJFSTX, 2023-08-03:
Hi realthunder,

I'm using the latest version of the script with mixed success on the via issues being resolved.

I am now getting an error that I've narrowed down to the via holes being made:

```
15:09:12  Traceback (most recent call last):
  File "C:/Users/HJFSTX/AppData/Roaming/FreeCAD/Macro/Import_PCB_Full.FCMacro", line 12, in <module>
    pcb.makeCoppers(shape_type = 'solid', holes = True, fuse = True, fit_arcs = True)
  File "C:\Users\HJFSTX\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad.py", line 2248, in makeCoppers
    drills = self.makeHoles(shape_type='solid',prefix=None,
  File "C:\Users\HJFSTX\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad.py", line 1534, in makeHoles
    w = make_circle(Vector(s))
  File "C:\Users\HJFSTX\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad.py", line 197, in make_circle
    return Part.Wire(Part.makeCircle(size.x*0.5))
<class 'Part.OCCError'>: creation of circle failed
```
If I remove the vias, the error goes away. I do have some oval PTH pads that will generate fine.

Many thanks and best regards.

qaqcvc, 2023-08-03:
> Hi realthunder,
>
> I'm using the latest version of the script with mixed success on the via issues being resolved.
>
> I am now getting an error that I've narrowed down to the via holes being made:
>
> ``` [...]

Hi HJFSTX, I also met this problem before. I then realized that the size of the via is too small in my case. The smallest size allowable in my board is 0.31,0.13 in my board. And this minimum size is limited by the thickness of the copper layer if I understand rightly, as the thickness of the via tube is proportional to the copper thickness.

You may try to change the size of the via to see if this issue persists.

HJFSTX, 2023-08-10:
@realthunder and @luzpaz 

I have been extensively using this script and am noticing what I am calling "phantom geometry" that is causing huge unforeseen issues in FEA. The geometry in question seems to have areas that are either shorted or totally unconnected, although they visually appear to be just fine. I think our issues are related.

realthunder, 2023-08-11:
Is it possible to provide a test file showing the problem?

HJFSTX, 2023-08-11:
@realthunder

I cannot post the exact source files, but if I encounter issues again I will attempt to replicate. Some of the issues I encountered were due to user error, so it may have been a false alarm.

HJFSTX, 2024-01-31:
@realthunder 

I'm again having some issues creating holes/vias. Please see below for the error:

`08:59:10  Traceback (most recent call last):
  File "C:/Users/HJFSTX/AppData/Roaming/FreeCAD/Macro/Import_PCB_Full_No_Holes_No_Zone_Inflate.FCMacro", line 12, in <module>
    pcb.makeCoppers(shape_type = 'solid', holes = False, fuse = True, fit_arcs = True)
  File "C:\Users\HJFSTX\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad.py", line 2239, in makeCoppers
    hole_coppers = self.makeHoles(shape_type='solid',prefix=None,
  File "C:\Users\HJFSTX\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad.py", line 1484, in makeHoles
    size = Vector(p.drill[0],p.drill[1])
  File "C:\Users\HJFSTX\AppData\Roaming\FreeCAD\Macro\fcad_pcb\kicad_parser\sexp_parser\sexp_parser.py", line 146, in __getitem__
    v = self._value[key]
<class 'KeyError'>: (1,)`

Any idea what's going on here?

qaqcvc, 2024-01-31:
Hi realthunder, thank you for your contribution on facd_pcb. I think your script to output real 3d layout is till now the only solution. I do not see any intention from Kicad to support this feature in the future. 

Just curious, would you consider to make a simple workbench for fcad_pcb, which has better UI and tuning options? The installation would be also easier. That would be tons of help for our FEM guys. 

BTW, I have shifted my freecad to your linkstage. Really like the features you developed. Thank you!

realthunder, 2024-02-03:
> I'm again having some issues creating holes/vias. Please see below for the error:

You seem to be using an older version of this repo. Could you please update it and try again?

realthunder, 2024-02-03:
> Just curious, would you consider to make a simple workbench for fcad_pcb, which has better UI and tuning options?

I may extend it into a FreeCAD macro with UI for settings. Not in the near future, though. Quite busy these days.

realthunder, 2024-02-03:
> Just curious, would you consider to make a simple workbench for fcad_pcb, which has better UI and tuning options?

Or, you can try the [kicadStepUp](https://github.com/easyw/kicadStepUpMod) workbench.

HJFSTX, 2024-02-05:
> > I'm again having some issues creating holes/vias. Please see below for the error:
> 
> You seem to be using an older version of this repo. Could you please update it and try again?

I pulled the latest version, tried again, and got the same result.

realthunder, 2024-02-05:
Could you please send me the file having problem?