realthunder/fcad_pcb #71: setNetFilter does not find the nets

tapegoji, 2024-05-13:
Hi, 
I just tried the setNetFilter function. However, it does not find the net, although it is in the nets. 

I used test.kicad_pcb board from your github

My commands:
from fcad_pcb import kicad
pcb = kicad.KicadFcad('./test.kicad_pcb')
pcb.setNetFilter('SDA1')

pcb.makeCopper()

Here is the error:
17:42:06  no KISYS3DMOD found
17:42:06  net SDA1
 not found
17:42:06  making copper layer F.Cu...
17:42:06    making pads...
""

[test.zip](https://github.com/realthunder/fcad_pcb/files/15300996/test.zip)

tapegoji, 2024-05-14:
found the issue. With the newer versions of kicad the nets are saved in "Net" format. this might be the reason.
Looking int the code I think if you change line #832 from
ndict[n[1]] = n[0]

to: 

ndict[n[1].strip('"')] = n[0]

andylinpersonal, 2024-11-16:
Seems like you should specify the net id (?) instead of the net name at least under Kicad v8.
[image]